stm32duino/Arduino_Tools #4: platform.txt in both F1 and L4 cores looks for the mac os x tools in dir 'macosx'

ProjectsByJRP, 2016-11-16:
rename directory from macos to macosx fixes it

rogerclarkmelbourne, 2016-11-17:
@ProjectsByJRP

This is unrelated, but I'm having some other issues on OSX ElCapitan.
The old bootloader does not seem to be recognised by ElCapitan.
The new bootloader is recognised but does not seem to get reset correctly after upload (by dfu-util)
There have been some changes to the bootloader to speed up the process between the upload finishing and the sketch running, which may be causing problems on the Mac, but I suspect that what I'm seeing is all to do with ElCapitan :-(

ProjectsByJRP, 2016-11-17:
I will help you any way I can but
I use Sierra [which probably has whatever bug/feature is causing your error]
I have not changed the boot loader as I’m using the HAL cores
let me know what you would like me to do
-john

rogerclarkmelbourne, 2016-11-17:
Hi

fpistm, 2017-08-14:
Close this one as no activities and link to old core.
Could be reopened when BP will be merged in the new core and if issue will be reproduced .

commit a0581df2 (2016-11-17): Fixed #4
closed 2017-08-14 by fpistm