stm32duino/Arduino_Tools #6: Unable to upload using massStorageCopy

fpistm, 2017-06-14:
_From @RickKimball on June 10, 2017 16:22_

I'm creating a new variant for the NUCLEO-F030R8. When using the massStoraryCopy build, platform.txt uses the {node} parameter of the board to decide where to copy the bin file at upload time. This fails for me when I follow the scheme used by all the other board.txt entries.

On my 64 bit linux 16.04, my NUCLEO-F030R8 board shows up with the name mass storage name of "NUCLEO" instead of "NODE_F030R8".

I tried running Win7 in a VM and it also shows up as "NUCLEO" there.  

Is there an expected driver or udev entry I'm supposed to be using? Or should the my board entry look like this:
```
Nucleo_64.menu.Nucleo_64_board.NUCLEO_F030R8.node=NUCLEO
```
BTW: that works properly

_Copied from original issue: stm32duino/Arduino_Core_STM32#23_

fpistm, 2017-06-14:
Yes you have to set it with NUCLEO as displayed by the host os. For other board this is also the name diplayed by host os.

fpistm, 2017-06-14:
As I have mentioned in other thread.
The mount point of my F030R8 is:
`/dev/sdc on /media/<login>/NODE_F030R8 type vfat (rw,nosuid,nodev,uid=xxxxxx,gid=xxxxxx,shortname=mixed,dmask=0077,utf8=1,showexec,flush,uhelper=udisks2)`

fpistm, 2017-06-14:
Depending of the board revision, mount point name could be different.
See this [thread](http://stm32duino.com/viewtopic.php?f=48&t=2187#p29615)
Ex for: STM32F030R8 with a rev 1 node name is : "NUCLEO" while it is "NODE_F030R8" for rev c

MassStorageCopy should be updated to manage several node names.
ex in boards.txt we could have:
Nucleo_64.menu.Nucleo_64_board.NUCLEO_F030R8.node="NODE_F030R8,NUCLEO"

fpistm, 2017-06-16:
Still MacOsx to build.

fprwi6labs, 2017-06-19:
I can build it for MacOSX.

commit 2d677b41 (2017-06-16): MassStorageCopy updated to manage several node names.

Depending of the board revision, mount point name could be different.
Ex for: STM32F030R8 with a rev 1 node name is : "NUCLEO"
while it is "NODE_F030R8" for rev c

Ex: in boards.txt we could have:
Nucleo_64.menu.Nucleo_64_board.NUCLEO_F030R8.node="NODE_F030R8,NUCLEO"

Fix #6
commit c01f5f2f (2017-06-16): MassStorageCopy updated to manage several node names.

Depending of the board revision, mount point name could be different.
Ex for: STM32F030R8 with a rev 1 node name is : "NUCLEO"
while it is "NODE_F030R8" for rev c

Ex: in boards.txt we could have:
Nucleo_64.menu.Nucleo_64_board.NUCLEO_F030R8.node="NODE_F030R8,NUCLEO"

Fix #6
commit 82eb13a6 (2017-06-16): MassStorageCopy updated to manage several node names.

Depending of the board revision, mount point name could be different.
Ex for: STM32F030R8 with a rev 1 node name is : "NUCLEO"
while it is "NODE_F030R8" for rev c

Ex: in boards.txt we could have:
Nucleo_64.menu.Nucleo_64_board.NUCLEO_F030R8.node="NODE_F030R8,NUCLEO"

Fix #6
PR #8 (2017-06-20): MassStorageCopy updated to manage several node names.
Depending of the board revision, mount point name could be different.
Ex for: STM32F030R8 with a rev 1 node name is : "NUCLEO"
while it is "NODE_F030R8" for rev c

Ex: in boards.txt we could have:
Nucleo_64.menu.Nucleo_64_board.NUCLEO_F030R8.node="NODE_F030R8,NUCLEO"

Fix #6
closed 2017-06-20 by fpistm