stm32duino/Arduino_Tools #17: genpinmap doesn't support stm32f10x chips

RickKimball, 2017-08-18:
When I try to use genpinmap with an stm32f103 board I get the message:

"STM32F10xx MCU not supported due to older GPIO IP version"

How did you generate the PeriphalPins.c used with the NUCELO_F103RB and DISCO_F100RB ?

fprwi6labs, 2017-08-18:
@RickKimball 

STM32F1 chips are not supported due to the alternate functions.
You have 2 ways to generate the PeripheralPins.c:
- hack genpinmap to force to generate the file with AF field empty (the one I used).
- generate manually the file from the existing.

I don't have better solutions to propose to you.

RickKimball, 2017-08-18:
Can you give me a hint how to hack it?

fpistm, 2017-08-18:
Hi @RickKimball 
I've push a branch on my forked repo with the hack: 
https://github.com/fpistm/Arduino_Tools/commit/b406f1c367a256faa18a520ae5f93751a622f614

RickKimball, 2017-08-18:
Thanks! That seems to work.

fpistm, 2017-08-18:
You're welcome.
I've updated my branch with a parsing for the F1 AF.
It required further study to align value btw core and __HAL_AFIO_REMAP_xxxxx:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/PinAF_STM32F1.h#L67

I think there are some issue in the current PeripheralPins.c used by our F1 variant.
This could be the root cause of the trouble with Serial Recv

RickKimball, 2017-08-18:
I wonder why the NUCLEO_F103RB works then?

fpistm, 2017-08-18:
If the concerned pins is not impacted it could work. It need further investigations. It's ongoing ;)

fpistm, 2017-08-18:
I will produce a patch to support all AF for F1 but this will request some core update that I need to validate.
The genpinmap script will be also updated to manage properly AF for F1.

fpistm, 2017-08-19:
@RickKimball,
here #18 , the PR to be able to extract AF information for F1.
I've also push a PR to the core to be able to handle all AF:
https://github.com/stm32duino/Arduino_Core_STM32/pull/84
If you can give a try with your vldisco, I've some doubt around some AF which could explain issue with Serial Rx
Thanks in advance

fpistm, 2017-08-21:
PR #18 merged.

closed 2017-08-21 by fpistm