stm32duino/Arduino_Tools #24: stlink_upload.bat does not work with sloeber v4.3 (solution here, Request for Comments)

swissbyte, 2018-08-16:
Hello

I have tried to use the stm32duino with sloeber. 
It seems that a lot of them works great. Expect for the upload tool. 

Sloeber calls the tool that way: 

`Launching: C:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/tools/win/stlink_upload.bat no_altID no_altID C:\code\sloeber\test_1_3_2/Release/test_1_3_2.bin`

the batch file tells me that it cant find the file. 

If i add 

`set str=%3`

after

`set str=%str:/=\%`

then it works like a charm. 

So is there any need to adjust the batch file or where does the failure happen?
Does sloeber call the batch file the wrong way? 

Thanks

fpistm, 2018-08-16:
Hi,
It seems that Sloeber do not call it as expected.
`tools.stlink_upload.upload.pattern="{path}/{cmd}" {serial.port.file} {upload.altID} {upload.usbID} "{build.path}/{build.project_name}.bin"`

It misses `{serial.port.file} ` that's why it is not ok. Have you select a COM port even if it is not linked to the board.

swissbyte, 2018-08-16:
Great! Thanks!

That was the problem

closed 2018-08-16 by swissbyte