stm32duino/Arduino_Tools #25: Sloeber expects the tools folder to be in a different folder

swissbyte, 2018-08-16:
Hi, 

Im currently testing sloeber with stm32duino. 
Unfortunatle sloeber tries to call the stlink tools at a wrong place: 

`Launching: C:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/tools/win/stlink_upload.bat`

As you can see here, sloeber expects the tools to be under hardware instead of tools. 

The correct path would be: 

`tools\STM32Tools\1.1.0\tools`

Where do we specify this path?
Is this a sloeber related problem or is it a misconfigured textfile from stm32duino?

Thanks

fpistm, 2018-08-16:
I guess you use the git repo and not the package one?
The platform.txt in the repo is not exactly the same than the one packaged.
In the git repo: 
`{runtime.hardware.path}`
is replace by:
`{runtime.tools.STM32Tools.path}`

This is mentioned here:
https://github.com/stm32duino/wiki/wiki/Using-git-repository#31-cloning-the-git-repository-to-replace-the-stm32-core-version-package-1st-method

swissbyte, 2018-08-16:
Interesting. 

I have added the package json in sloeber and let them download all files by them self.

fpistm, 2018-08-16:
So, you should not get this issue if you use only the package.
Did you check the platform.txt ?
You should have:
`tools.serial_upload.path={runtime.tools.STM32Tools.path}/tools/win`

swissbyte, 2018-08-16:
ok i got the problem. 
I have messed up something during developing code for the generic STM32F0. 
It was a problem with the git repo. as you already told me. 

Sorry for the inconvenience.

closed 2018-08-16 by swissbyte