stm32duino/Arduino_Tools #28: Can dfu-util be updated?

chrissbarr, 2018-10-04:
I might be wrong, but it looks like the included dfu-util tool (at least on Windows) is quite an old version.

I can't get it to detect any STM32F4 microcontroller in DFU mode - either running the built-in bootloader or my own DFU bootloader in flash. I haven't tried with any other STM family yet.

More recent versions of dfu-util seem to work fine with these boards. I tested with v0.9, which seems to be the latest, and was able to get it to work by dropping it into the tools/win directory (along with the libusb-1.0.dll it is packaged with). This makes it simple to add dfu-util as an upload method in the Arduino_Core_STM32, and seems to work for me with chips that have the DFU bootloader built into them.

Is there any reason not to do this? Or should the included dfu-util be working as-is - is anyone else using it successfully?

If it's just a matter of dropping in the updated dfu-util, I can put together a pull request. It just seems too straightforward, so I figure either there's a reason not to do it, or there's more to it than I'm seeing.

fpistm, 2018-10-04:
Hi @chrissbarr 
The only reason is that the time is always a missing ingredient. Feel free to provide a PR for that.
I will release soon a new version, this should be fine to include this update.

fpistm, 2019-02-13:
Fix thanks #29

closed 2019-02-13 by fpistm