stm32duino/Arduino_Tools #31: STM32F030C8 pinmap is missinf PF6 and PF7 as I2C pins

ghent360, 2018-12-26:
Hi,

The generated pinmap for the STM32F030C8 chip has only PB pins as available for I2C. In the STM32CubeMX software I can select pins 35 and 36 (PF6 and PF7) as SCL and SDA for the I2C2 module.

The generated code for the GPIO init is:
    __HAL_RCC_GPIOF_CLK_ENABLE();
    /**I2C2 GPIO Configuration    
    PF6     ------> I2C2_SCL
    PF7     ------> I2C2_SDA 
    */
    GPIO_InitStruct.Pin = GPIO_PIN_6|GPIO_PIN_7;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_OD;
    GPIO_InitStruct.Pull = GPIO_PULLUP;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
    HAL_GPIO_Init(GPIOF, &GPIO_InitStruct);

    /* I2C2 clock enable */
    __HAL_RCC_I2C2_CLK_ENABLE();

It is a bit odd, because there is no GPIO_InitStruct.Alternate. There is also no mention for PF6 and PF7 in the datasheet :-(.

fpistm, 2018-12-27:
From datasheet: https://www.st.com/resource/en/datasheet/stm32f030c8.pdf

[image]
[image]

The genpinmap script used to generate PeripheralPins.c parse xml from STM32CubeMX.
In the xml:
```
	<Pin Name="PF6" Position="35" Type="I/O">
		<Signal Name="I2C2_SCL"/>
		<Signal IOModes="Input,Output,Analog,EXTI" Name="GPIO"/>
	</Pin>
	<Pin Name="PF7" Position="36" Type="I/O">
		<Signal Name="I2C2_SDA"/>
		<Signal IOModes="Input,Output,Analog,EXTI" Name="GPIO"/>
	</Pin>
```

But not AF linked to this in GPIO xml file, that's why they are not present. When no AF is found the pin is ignored. So there is an issue with thoses pins definition, because I guess with CubeMX code this is not working? Did you try to use those pins for I2C ? 
I saw in your PR that you added them manually with an AF? Are they working ?

If yes, the best way is to mention that it was added manually in the variant.
The `PeripheralPins.c` are generated so while the xml are not corrected this could not be corrected in those files. I will raise internally to see how to handle this properly.

fpistm, 2019-01-04:
@ghent360 
I've made an update of the script, now all pins are referenced. If no AF is linked then set GPIO_AFIO_NONE is set.

PR #32 (2019-01-07): Add missing pins
Fix #31
commit 840b8d62 (2019-01-07): [genpinmap] Add missing pins

Fix #31