stm32duino/Arduino_Tools #41: Mass Storage upload "NODE not found error"

MauroMombelli, 2019-09-30:
If i connect my nucleo to the pc i see the nuvleo recognized as mass storage device.
If i try to program it, i get the error:

```
An error occurred while uploading the sketch
NODE_F303RE not found. Please ensure the device is correctly connected
```
I believe that is caused by the fact that the device is seen (lsusb) but does not get automatically mounted (default in kde 5, but for what i find around seems more a systemd stuff)

a quick fix is to ask the user to check if the drive is mounted (it is done once the user try to open it)

a proper fix would be to check if the disk is mounted, and if not it can be mounted/unmounted with 'udisksctl` for example: 'udisksctl mount -b /dev/sde /tmp/nucleo`

fpistm, 2019-09-30:
I don't think this is the script which need to perform the mount.
This is to user to do this as it is related to their (system) rules moreover it could required admin rights.

MauroMombelli, 2019-10-01:
> it could required admin rights

this is why `udisksctl` is used

> it is related to their (system) rules

this is the default kde behavior, for what i can find since 2010, see https://www.maketecheasier.com/manage-removable-devices-in-kde/

> To the right of the device information, you will see an icon that looks like a plug or connector. Clicking it will immediately mount it but will not perform any further actions. This is useful if you want it mounted but do not necessarily want to access it at the moment.

fpistm, 2019-10-01:
> good with me, would be nice to simply have a better error message then

Do not hesitate to provide a PR with better description, this is a community project, any contribution are welcome.

matthijskooijman, 2019-10-11:
IMHO, automatically mounting the device would make sense and certainly make things easier to work with. I'm not sure how easy and portable the udiskctl command is, IIRC there have been quite some different ways how mounting works over the years, some of them desktop-environment-specific, but maybe this somewhat stabilized by now? If there is one somewhat-uniform mounting command, using that would make sense (it would then improve things for a lot of users and keep things unchanged for anyone using some other system)?

fpistm, 2019-10-11:
I will not add this kind of feature. It is the responsibility of the user to mount a device or configure their rules to automatically mount or not. Moreover, this could rely on some specific commands related to security (ICT rules, ...).

PR #43 (2019-10-14): Linux64 removal
No more binary issue due to host arch (32 or 64 bits).
Message updated:

> Please ensure the device is correctly connected and mounted.

This closes #41 and #42.
commit 4f8802e8 (2019-10-14): massStorageCopy becomes a script

This closes #41 and #42.
closed 2019-10-14 by fpistm