stm32duino/Arduino_Tools #42: Mass storage upload fails on linux64 when no 32-bit loader is installed

matthijskooijman, 2019-10-10:
On a 64-bit Ubuntu Disco system, mass-storage uploads fail. In the IDE window, I get:

    java.io.IOException: Cannot run program "/home/matthijs/.arduino15/packages/STM32/tools/STM32Tools/1.3.1/tools/linux/massStorageCopy": error=2, No such file or directory

Closer inspection shows that the file referenced actually does exist, but is a 32-bit binary. Executing it fails with a `ENOENT`, which is expected (according to https://unix.stackexchange.com/a/13409/233460) when no 32-bit loader is installed.

Manually starting the binary also fails, and strace shows it is the actual `execve` systemcall that fails:

    execve("/home/matthijs/.arduino15/packages/STM32/tools/STM32Tools/1.3.1/tools/linux/massStorageCopy", ["/home/matthijs/.arduino15/packages/STM32/tools/STM32Tools/1.3.1/tools/linux/massStorageCopy"], 0x7ffe812fecd8 /* 95 vars */) = -1 ENOENT (No such file or directory)

A linux64 version is included, which does run and work as expected (tested by editing `platform.txt`).

Looking at the `platform.txt`, at first  glance it looks like there is a typo in the linux64 version: https://github.com/stm32duino/Arduino_Core_STM32/blob/6808f06a3c183f22784f0486a3034644535c5b2b/platform.txt#L160-L161

As you can see, both refer to the 32-bit version. However, on closer inspection, it turns out that the Arduino IDE does not actually [distinguish between linux32 and linux64 currently](https://github.com/arduino/Arduino/blob/1.8.10/arduino-core/src/processing/app/helpers/PreferencesMap.java#L110-L112), so to really fix this, I guess the IDE needs to be changed (I'll file a bug report for that as well).

As a workaround, you can enable execution of 32-bit binaries on a 64-bit Linux Debian-derived (e.g. Ubuntu) system by enabling multiarch and installing a 32-bit libc:

    dpkg --add-architecture i386
    apt-get update
    apt-get install libc6:i386

With that, things work as expected again. But ideally, you would use the 64-bit version on a 64-bit system, if Arduino adds support for that.

**To Reproduce**
 1. Get a 64-bit system without 32-bit compatibility enabled (default on Ubuntu, it seems).
 2. Get any sketch, and select any STM32 board (e.g. Nucleo63 -> F401RE)
 3. Select upload method mass storage
 4. Press upload

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu Disco
 - Arduino IDE version: Git master (a1448876a1115c9d3ee9e88f29a15bb081a27816), bit newer than 1.8.11
 - STM32 core version: 1.7.0
 - Upload method: Mass Storage

**Board (please complete the following information):**
 - Name: Nucleo F401RE

matthijskooijman, 2019-10-11:
Ah, I see the method you proposed. It is really similar to what I suggested, except that you delete the files from ~/.arduino15 rather than rename the core from git in the sketchbook.

However, with "your" method, having a split tools tarball also still works, since the tools are installed by the IDE as well.

Previously, you said:

> Thinking about that, there will be always an issue for those using the git repo which include all tools.
This will not solve the pb.

I think this refers to using a modified Arduino_Tools repository, not a modified Arduino_Core_STM32 repository. And for doing that, I think you need to modify `platform.txt` anyway, is what I was saying.

fpistm, 2019-10-11:
> However, with "your" method, having a split tools tarball also still works, since the tools are installed by the IDE as well.

Rigth

> I think this refers to using a modified Arduino_Tools repository, not a modified Arduino_Core_STM32 repository. And for doing that, I think you need to modify `platform.txt` anyway, is what I was saying.

No, I delete the tools directory then clone the Arduino_Tools repo so no need to update platform.txt.
Anyway, as I said there is lot of method for use git repo, this is finally up to end user to know what he does.

fpistm, 2019-10-11:
Arduino provide this feature and I do not want have manually create a symlink to use the git repo. This should work as it is.
I'm currently removing the `linux64` directory then no more needs about all that stuff.

matthijskooijman, 2019-10-11:
> I'm currently removing the linux64 directory then no more needs about all that stuff.

So that means always using the 32-bit binaries, also on 64-bit linux? That requires people to install additional libraries, which might not be so easy for everyone (e.g on Debian it is more than just installing an extra package). Also, that would not work if you ever want to support arm rather than x86 (which the Arduino IDE already does).

fpistm, 2019-10-11:
No, I didn't tell that.
As I said I will provide a bash script instead of a binary.
And for other binary I've updated the bash script to call the correct binary depending of the host arch (i.e. dfu-utils).

matthijskooijman, 2019-10-11:
Ok, then I misunderstood. I guess a script would work, but it doesn't seem too elegant to duplicate the arch detection again in a script. But it does make it easier to run with tools from git, I guess, so if that's what you prefer :-)

Thanks!

fpistm, 2019-10-11:
Well, this avoid to duplicate files in the repo and have it functional without have to add a platform.local.txt, or create a symlink. 

Only one arch detection have to be performed:
```
if [ `uname -m` ==  "x86_64" ]; then
  DFU_UTIL=${DIR}/dfu-util_x86_64/dfu-util
else
  DFU_UTIL=${DIR}/dfu-util/dfu-util
fi
```
I guess this is not very disturbing

fpistm, 2019-10-14:
Closed by #43

closed 2019-10-14 by fpistm