stm32duino/Arduino_Tools #45: [dfu-util] i686 files don't run on 32-bit systems

jgillick, 2019-10-20:
_First off, I apologize if I am misunderstanding the structure of linux/linux_x86_64_. 

From what I can tell, the `linux` directory is for i686 systems (32-bit). However, when I was just trying to test it in a 32-bit virtualized environment, it seems that both versions of `dfu-util` are compiled for 64-bit systems and will not execute on 32-bit.

I was running this in Ubuntu 16.04, 32-bit, in VMWare Fusion on my mac.

**Error:**

```bash
$ ./linux/dfu-util/dfu-util
bash: ./linux/dfu-util/dfu-util: cannot execute binary file: Exec format error
```

**Relevant output:**

```bash
$ uname -m
i686

$ file ./linux/dfu-util/dfu-util
./linux/dfu-util/dfu-util: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib, for GNU/Linux 2.6.32, BuildID[sha1]=c58354ce3ca661a39bc97aacb5beffa73cf5dfb3, not stripped

$ file ./linux/dfu-util_x86_64/dfu-util
./linux/dfu-util_x86_64/dfu-util: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib, for GNU/Linux 2.6.32, BuildID[sha1]=683c50b08daafbb5db685fc3d9503d0c160fe108, not stripped
```

fpistm, 2019-10-21:
Hi @jgillick 
Right, I've missed that when refactoring.
In fact dfu-util binaries came from legacy core for Roger.
I didn't check arch and assuming thy are aligned (linux: 32 bits and linux64: 64 bits).
To solve this the 32 bits version needs to be provided.

jgillick, 2019-10-21:
I could compile the 32-bit version in the virtualized environment I have and provide a PR if that would be helpful.

fpistm, 2019-10-21:
It would be helpful, like this you could test it.

jgillick, 2019-10-21:
Good news. I just confirmed that the official `i386` binaries, linked from the dfu-util site (http://dfu-util.sourceforge.net/releases/dfu-util-0.8-binaries.tar.xz), work for 32-bit systems. I'm adding this to my existing PR: https://github.com/stm32duino/Arduino_Tools/pull/46

fpistm, 2019-10-21:
Thanks a lot @jgillick 
I will review that soon.

closed 2019-12-19 by fpistm