stm32duino/Arduino_Tools #51: how to pass ST-Link serial number to stm32programmer ?

jmchiappa, 2020-01-20:
Hello,
When several NUCLEO boards are connected, how can  I specify to stm32programmer the expected serial number to flash with ?
In other words, STM32Programmer CLI can receive a "sn" option that specifies the serial number, but I don't find any placeholder in boards.txt to send it to.

fpistm, 2020-01-20:
Hi @jmchiappa 
currently this is not supported.
The main issue is how to handle the Serial Number as Arduino has no way to have user input (AFAIK).

jmchiappa, 2020-01-20:
I agree.
I'm working on web application that sends some codes to local server. it wraps the STM32duino framework and uses its toolchain.
Local server is written in NodeJS, and, fortunately, serialport object enumerates usb device with its full descriptor and, by the way, the serial number. So,  it's really easy in this case to select from a dropdown menu the right board and therefore the right serial number.
Again, I agree that it's pretty far from the official use from Arduino IDE. But this framework is really versatile and pretty easy to embed in third party IDE.

In fact, I just need a placeholder to drop the sn that will be forwarded to STM32programmer.
So any idea is welcome !

fpistm, 2020-01-20:
Then you should add a new variable in the platform.txt and update the stm32programmer  cube programmer to use it for your needs.
As well the boards.txt probably needs to be updated to create your own menu.
Anyway this could be apply only for your usage. Maybe use the platform.local.txt to override the default one could be a good option.

jmchiappa, 2020-01-20:
ok, i will have a try... thank you

closed 2020-01-24 by jmchiappa

devel0, 2020-10-25:
I definitively needed to manage such thing in a hugly but safe way in order to avoid a sketch to upload on wrong target when more than a board connected, so to avoid `arduino upload` from vscode to target my nucleo64 instead of the minimaple stm32f108 I used follow changes [see here](https://github.com/devel0/repros/commit/7c8bb60948a74830182b7c75daae644c16e25ab3).

then I choosen it from Arduino Board Configuration

[image]

In other word I added a static menu entry with my custom sn board ( I don't know if I could invoke some script shell from inside boards.txt to parse available serial numbers from `STM32_Programmer_CLI -l` ) then I needed to change in a backward compatible way the `STM32_Programmer.sh` to revert any `SWDSNxxx` port to 0 setting back PORT to swd but saving serial number info to pass into sn=xxx arg.