stm32duino/Arduino_Tools #57: [STM32CubeProg] Add an offset for upload?

GIPdA, 2020-03-05:
Hello

I'm using the STM32CubeProgrammer to upload, and I just had the use of an address offset, to upload code via Arduino without overwriting my custom bootloader.

I modified the upload script for my purposes, but it may be useful to add it properly as a feature/option? (and I wouldn't need to modify everything again after updates :P )

I added this to the upload script:
```
case $1 in
    -o|--offset)
    OFFSET=$2
    ADDRESS=$((ADDRESS+OFFSET))
    shift # past argument
    shift # past value
    ;;
esac
```

And to the boards.txt file:
```
GenF3.menu.pnum.MY_BOARD.upload.extra_options=-o 0x2800
...
GenF3.menu.upload_method.swdMethod.upload.options={upload.extra_options} -g
```

I'm no expert in scripts or Arduino's board definitions syntax, maybe there's a better way?
I can add a pull request if desired and if the way I did it seems suited. What do you think?

Cheers,
Benjamin

fpistm, 2020-03-05:
Hi @GIPdA 
yes this would be fine to add this feature.
I think about this but do not add it when I moved to the STM32CubeProgrammer as there is a builtin bootloader which handle most case so no need to have an offset.
Just for my interest what is the goal of your bootloader ?

GIPdA, 2020-03-05:
Hi @fpistm 
Great! I will see to do a pull request soon then. And I just though of that but I will need to do a second one on the core repo too, where the board.txt is.
It is a simple bootloader over CAN, for a STM32F303V.

ricaun, 2020-03-31:
Hello, @fpistm and @GIPdA 

This offset feature should be incredibly good to protect de bootloader and upload the code in the right address.

But I think the best way is adding a new parameter on the `stm32CubeProg` using the already implemented `build.flash_offset`. 

Doing that this board configuration gonna upload on the `0x08005000` using the `stm32CubeProg` tool.
```
GenF1.menu.upload_method.serialMethod2=STM32CubeProgrammer (Serial) - Bootloader
GenF1.menu.upload_method.serialMethod2.upload.protocol=1
GenF1.menu.upload_method.serialMethod2.upload.options={serial.port.file} -s
GenF1.menu.upload_method.serialMethod2.upload.tool=stm32CubeProg
GenF1.menu.upload_method.serialMethod2.build.flash_offset=0x5000
GenF1.menu.upload_method.serialMethod2.build.bootloader_flags=-DVECT_TAB_OFFSET={build.flash_offset}
```

To make that happen we need to change the `platform.txt` on the `Arduino_Core_STM32` and change the line.

```
tools.stm32CubeProg.upload.pattern="{path}/{cmd}" {upload.protocol} "{build.path}/{build.project_name}.bin" {build.flash_offset} {upload.options}
```

And change the `stm32CubeProg.bat` on the `Arduino_Tools` like:

```
@echo off

set ERROR=0
set STM32CP_CLI=STM32_Programmer_CLI.exe
set ADDRESS=0x8000000
set ERASE=
set MODE=
set PORT=
set OPTS=

:: Check tool
where /Q %STM32CP_CLI%
if %errorlevel%==0 goto :param
::Check with default path
set STM32CP=%ProgramW6432%\STMicroelectronics\STM32Cube\STM32CubeProgrammer\bin
set STM32CP86=%ProgramFiles(X86)%\STMicroelectronics\STM32Cube\STM32CubeProgrammer\bin
set PATH=%PATH%;%STM32CP%;%STM32CP86%
where /Q %STM32CP_CLI%
if %errorlevel%==0 goto :param
echo %STM32CP_CLI% not found.
echo Please install it or add ^<STM32CubeProgrammer path^>\bin' to your PATH environment:
echo https://www.st.com/en/development-tools/stm32cubeprog.html
echo Aborting!
exit 1

:param
:: Parse options
if "%~1"=="" echo Not enough arguments! & set ERROR=2 & goto :usage
if "%~2"=="" echo Not enough arguments! & set ERROR=2 & goto :usage
if "%~3"=="" echo Not enough arguments! & set ERROR=2 & goto :usage

set PROTOCOL=%~1
set FILEPATH=%~2
set OFFSET=%~3

:: Add offset on address
SET /A ADDOFF=%ADDRESS%+%OFFSET%
cmd /C exit %ADDOFF%
set "ADDRESS=0x%=ExitCode%"

:: Protocol
:: 1x: Erase all sectors
if %~1 lss 10 goto :proto
set ERASE=-e all
set /a PROTOCOL-=10

:: 0: SWD
:: 1: Serial
:: 2: DFU
:proto
if %PROTOCOL%==0 goto :SWD
if %PROTOCOL%==1 goto :SERIAL
if %PROTOCOL%==2 goto :DFU
echo Protocol unknown!
set ERROR=4
goto :usage

:SWD
set PORT=SWD
set MODE=mode=UR
goto :opt

:SERIAL
if "%~4"=="" set ERROR=3 & goto :usage
set PORT=%~4
shift
goto :opt

:DFU
set PORT=USB1
goto :opt

:opt
shift
shift
shift
if "%~1"=="" goto :prog
set OPTS=%1 %2 %3 %4 %5 %6 %7 %8 %9
goto :prog

:prog
%STM32CP_CLI% -c port=%PORT% %MODE% %ERASE% -q -d %FILEPATH% %ADDRESS% %OPTS%
exit 0

:usage
  echo %0 ^<protocol^> ^<file_path^> [OPTIONS]
  echo.
  echo protocol:
  echo   0: SWD
  echo   1: Serial
  echo   2: DFU
  echo    Note: prefix it by 1 to erase all sectors
  echo          Ex: 10 erase all sectors using SWD interface
  echo file_path: file path name to be downloaded: (bin, hex)
  echo Offset:
  echo   Address offset
  echo Options:
  echo   For SWD and DFU: no mandatory options
  echo   For Serial: ^<com_port^>
  echo     com_port: serial identifier (mandatory). Ex: COM15
  echo.
  echo Note: all trailing arguments will be passed to the  %STM32CP_CLI%
  echo   They have to be valid commands for STM32 MCU
  echo   Ex: -g: Run the code at the specified address
  echo       -rst: Reset system
  echo       -s: start automatically (optional)
  exit %ERROR%
```

I don't know how to change the `linux` and `macosx` part 😞 

I gonna create a pull request, what do you think?!

PR #90 (2023-04-05): feat(stm32CubeProg): add offset option
Fixes #57.
Replace #60.
Requires update in the core boards.txt and platform.txt: https://github.com/stm32duino/Arduino_Core_STM32/pull/1993
commit c2b8d22e (2023-04-05): feat(stm32CubeProg): add offset option

Fixes #57
closed 2023-04-05 by fpistm