stm32duino/Arduino_Tools #58: MacOS Catalina PORT='USB1' -> PORT='usb1'

rayozzie, 2020-03-08:
Hello.  I am trying to incorporate the Adafruit Feather STM32F405 Express into the design of a dev kit that I'm building.  Out-of-the-box Arduino support is critical.

I followed install instructions, and on MacOS the stm32duino scripts fail when trying to program the board.  Here is the support history on the Adafruit site, fyi:  [](https://forums.adafruit.com/viewtopic.php?f=57&t=163183)

I reduced the problem down to the fact that, for some reason, the script in macosx/stm32CubeProg (which is called macosx/stm32CubeProg.sh in this repo) has a problem.  (I do not understand why it has a .sh extension in this repo but it has no extension in the version packaged with stm32duino.)

As you can see in the above forum post, I ultimately found that the 'fix' was to change PORT='USB1' to PORT='usb1' in the macosx script.

I would deeply appreciate your fixing and deploying this asap because it'll be difficult to get users to find and edit this file themselves, and the failure mode will be somewhat mysterious to them.  Thank you.

fpistm, 2020-03-09:
Hi @rayozzie 

> As you can see in the above forum post

Which forum  post?

About the script extension, this is normal as it was renamed to unified the name, see
https://github.com/stm32duino/Arduino_Tools/pull/56

About your issue, I've tested with a Mac (also with Catalania version) and an Adafruit Feather STM32F405 Express but I did not reproduced your issue with "USB1".

I've also tested with "usb1" and this works if the STM32CubeProgrammer is a version higher than 2.1.0 else it's failed. The latest version (2.4.0) allows to use both syntax.

So I'm confused about your issue. Which STM32CubeProgrammer you used ?

rayozzie, 2020-03-09:
Well, this is interesting and it is complicated.

I was using a very old STM32CubeProgrammer (1.2.0).

Unfortunately, the latest version appears to be broken and does not seem to work on MacOS Catalina 10.15.3

When I first try to install it, I get the problem. below because apparently your code is not digitally signed.  

However, even after I go into Security settings and allow it to "open anyway", when I try to open SetupSTM32CubeProgrammer-2.4.0.app, MacOS won't run it and I get:

LSOpenURLsWithRole() failed with error -10810 for the file /Users/rozzie/Desktop/en.stm32cubeprog_v2-4-0/SetupSTM32CubeProgrammer-2.4.0.app.

[image]

fpistm, 2020-03-09:
> Unfortunately, the latest version appears to be broken and does not seem to work on MacOS Catalina 10.15.3

I had the same issue than yuor screenshot anyway I succeed to install it by following the MacOS help. But I had to open it thanks the command line not from finder.

fpistm, 2020-03-09:
Did you try the method describe in the User Manual:
https://www.st.com/content/ccc/resource/technical/document/user_manual/group0/76/3e/bd/0d/cf/4d/45/25/DM00403500/files/DM00403500.pdf/jcr:content/translations/en.DM00403500.pdf

[image]

rayozzie, 2020-03-09:
This is not what the user interface looks like on Catalina.  What you see on Catalina is what you see below.  As you can see, I have it set to allow "identified developers".  Unfortunately, ST has not apparently had its code digitally signed.

(I've seen that there is apparently a "sudo spctl --master-disable" hack that can be used to remove security from MacOS, but neither I nor ST can reasonably ask users to do this.)

[image]

rayozzie, 2020-03-09:
By the way, even after I do the "sudo spctl --master-disable" and allowing apps to be downloaded from Anywhere, I am getting the following when I try to open it from the command line:

$ open SetupSTM32CubeProgrammer-2.4.0.app
LSOpenURLsWithRole() failed with error -10810 for the file /Users/rozzie/.Trash/en.stm32cubeprog_v2-4-0 7.21.58 AM/SetupSTM32CubeProgrammer-2.4.0.app.

rayozzie, 2020-03-09:
I am sorry - I was reading on a phone and didn’t see that the “master disable” hack is actually in the manual and the screen cap you posted.  I still think it is unreasonable to ask users to use terminal and disable Mac security to install an app.

That said, as noted above even that didn’t work for me.

fpistm, 2020-03-09:
On my side I did not use the `sudo spctl --master-disable`. I've just follow the help documentation available here:
https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unidentified-developer-mh40616/10.15/mac/10.15.2

Then I simply launch in the command line with the path to the Setup:
<your absolute or relative path>/SetupSTM32CubeProgrammer-2.4.0.app/Contents/MacOs/SetupSTM32CubeProgrammer-2_4_0_macos
Then I was able to install it.

I'm agreed this should works without all those stuff and this will need a better management. But I could not help on this stuff as I'm also an user 😞
I guess your best bet it to open a topic on this subject on https://community.st.com/ to get appropriate help.

rayozzie, 2020-03-09:
It’s unfortunate that I am not able to complete the install of 2.4.0 under 10.15.3, while you are able to do so.

I will open a topic there.

I will also open a topic related to the need for downloading a CLI version of the programmer without registering for the website.

Thanks, Frederic.

https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unidentified-developer-mh40616/10.15/mac/10.15.2

fpistm, 2020-03-09:
On my side I will check if dfu-utils could be integrated to by pass this.

rayozzie, 2020-03-09:
That would be tremendous, Frederic.  I am faced with the possibility that if it is not seamless and easy for users to download Arduino code and run it, they will (understandably) choose another Feather MCU.  Sigh.

fpistm, 2020-03-09:
For reference:
https://community.st.com/s/question/0D50X00009cdVnZSAU/stm32cubeprogrammer-fails-to-install-in-macos

rayozzie, 2020-03-09:
Well that explains it – thank you.

But my goodness, this is another example of something that will be impossible for ‘normal humans’ to figure out.

MacOS Catalina should be treated like a first-class platform in terms of ease-of-use, and right now it is clearly not the dev environment that your own devs must be using.

https://community.st.com/s/question/0D50X00009cdVnZSAU/stm32cubeprogrammer-fails-to-install-in-macos

fpistm, 2020-05-15:
I close this issue as it mainly related to STM32CubeProgrammer installation under MacOS.
dfu-utils usage will come in further release.

closed 2020-05-15 by fpistm

Briangsouza, 2020-06-25:
https://www.youtube.com/watch?v=NuYNmxmJKjU