stm32duino/Arduino_Tools #59: STM32CubeProgrammer manual install?

aster94, 2020-03-15:
Hello,

For my own curiosity may I know why the STM 32cubeprogrammer is not installed automatically but manually?

I can't see any pro in this but for sure I am wrong

fpistm, 2020-03-16:
@aster94 
The main reason is it could not be provided due to license agreement.
Moreover:
* It is an optional upload method as it exists some other
* User can also already  have it installed so no need to install it twice
* The tools is updated regularly, so it is easier to update it locally

Please for question use the forum: https://www.stm32duino.com

closed 2020-03-16 by fpistm

fpistm, 2020-03-16:
https://github.com/platformio/platform-ststm32/issues/356

aster94, 2020-03-16:
Great, thanks for the explaination, in the future I will use the forum