stm32duino/Arduino_Tools #63: STM32_Programmer.sh not found

bsamadi, 2020-05-02:
`STM32_Programmer.sh` is in `/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin` and this folder is added to the path. I can run `STM32_Programmer.sh` in the terminal with no issues. 

However, Arduino IDE complains that `STM32_Programmer.sh` was not found.

When I checked, the PATH inside `stm32CubeProg.sh` is different from the one in the terminal.

I even changed `$HOME` to `/usr/local` in `stm32CubeProg.sh` but the error is still the same.

fpistm, 2020-05-02:
Seems strange.
Which version of Arduino IDE  you used?
Coud you echo the $PATH in the stm32CubeProg.sh

bsamadi, 2020-05-02:
version: 1.8.12
PATH in `stm32CubeProg.sh`: 
`/snap/arduino/14/java/bin:/snap/arduino/14/usr/sbin:/snap/arduino/14/usr/bin:/snap/arduino/14/sbin:/snap/arduino/14/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games`

fpistm, 2020-05-02:
So STM32cubeprogrammer is not in the path.
How you launch the IDE?

bsamadi, 2020-05-02:
The PATH that I showed is at the beginning of `stm32CubeProg.sh`, coming from the echo $PATH command in this picture: http://i.imgur.com/N8jxZeB.png

One issue is that this PATH is different from the PATH in the user environment. 
The other issue is that I modified `stm32CubeProg.sh` (see the above picture) and add the installation folder of `STM32_Programmer.sh` to the PATH. However, I still get this error: http://i.imgur.com/J8v5hPp.png

fpistm, 2020-05-02:
Try to launch arduino from the terminal.

fpistm, 2020-05-02:
From:
https://help.ubuntu.com/community/EnvironmentVariables
> Launching desktop application with an environment variable
> You can add an environment variable to an application by editing its .desktop file. For example, to run "digiKam" with the environment variable APPMENU_DISPLAY_BOTH=1, find the corresponding digikam.desktop file and add the setting of the variable, via the env command, to the entry "Exec":
> 
> 
> Exec=env APPMENU_DISPLAY_BOTH=1 digikam -caption "%c" %i

Lauching from desktop application has a different env.
That's why it's not work. So add it as explained.

bsamadi, 2020-05-02:
We are talking about PATH not general environment variables. It seems that the issue originates from the fact that Ubuntu install Arduino IDE using snap. This is how PATH is defined in snap (https://snapcraft.io/docs/environment-variables):
[image]

fpistm, 2020-05-02:
Ok , anyway you need to have the cube programmer in the path. There is no bug. This works as expected and warn user that it cannot be found.

bsamadi, 2020-05-02:
It would be great if a new user could follow these steps to use STM32duino:

-  install Arduino IDE through `Software` in Ubuntu
- install [STM32CubeProg](https://www.st.com/en/development-tools/stm32cubeprog.html) in its default location (`/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer`)
- follow the instructions in [Getting Started](https://github.com/stm32duino/wiki/wiki/Getting-Started)

Currently, there are two issues with this workflow:

1. Ubuntu uses snap to install Arduino IDE and snap does not use the default user PATH
2. If `stm32CubeProg.sh` doesn't find `STM32_Programmer.sh` in PATH, it looks for it in `$HOME/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin` not the default installation path, which is: `/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin`.

fpistm, 2020-05-04:
@bsamadi 

>     * install [STM32CubeProg](https://www.st.com/en/development-tools/stm32cubeprog.html) in its default location (`/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer`)

As far as I can tell is  this is not the default location.

> Ubuntu uses snap to install Arduino IDE and snap does not use the default user PATH

Yes, so up to end user to provide the correct environment.

>     2\. If `stm32CubeProg.sh` doesn't find `STM32_Programmer.sh` in PATH, it looks for it in `$HOME/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin` not the default installation path, which is: `/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin`.

Yes, as it is the default path as far as I can tell.

How did you install the CubeProgrammer?

bsamadi, 2020-05-04:
@fpistm, I checked again. You are absolutely right. I had changed the default installation folder. STM32duino works just fine. My main problem was the Arduino IDE installed by snap.

Thank you!

closed 2020-05-04 by bsamadi

fpistm, 2020-05-04:
Welcome.
Thanks to have checked again.
About snap it is fine to be aware of this issue.

Diatrix, 2020-05-15:
@bsamadi What was your solution then? Uninstall the Arduino IDE and reinstall it though command line?

bsamadi, 2020-05-16:
Yes, I followed the instructions [here](https://www.arduino.cc/en/guide/linux).

ggatis, 2021-08-17:
Or change file stm32CubeProg.sh:
line
    export PATH="$HOME/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin":$PATH
to
    export PATH="/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin":$PATH

blaisegassend, 2024-02-11:
I noticed that when running from arduino in a snap, HOME is set to:
/home/my_user_name/snap/arduino/85

By installing STMicroelectronics into /home/my_user_name/snap/arduino/85 instead of /home/my_user_name, the snap version of arduino was able to find it just fine.

You can probably do this automatically by running
snap run --shell arduino
and running SetupSTM32CubeProgrammer-2.15.0.linux from there. Or you can just install normally and copy/move.

rtek1000, 2025-01-27:
The problem is caused when installing SetupSTM32CubeProgrammer-x.xx.x.linux using sudo or root. If installed without elevation, it will be installed in the user's home.

Wrong:

> sudo ./SetupSTM32CubeProgrammer-2.13.0.linux

`Path: /usr/local/STMicroelectronics/STM32Cube/`

Correct:

> ./SetupSTM32CubeProgrammer-2.13.0.linux

`Path: /home/user/STMicroelectronics/STM32Cube/`