stm32duino/Arduino_Tools #64: STM32_Programmer.sh not found in Ubuntu 18.04. Edited PATH, and still can't find it.

Diatrix, 2020-05-18:
Hello, 
I'm having issues uploading code to my STM32F103C8 board. When I try to upload the file, it throws out this error: 

> STM32_Programmer.sh not found.
>An error occurred while uploading the sketch
>Please install it or add '<STM32CubeProgrammer path>/bin' to your PATH environment:
>https://www.st.com/en/development-tools/stm32cubeprog.html
>Aborting!

I installed the STM32CubeProgrammer, to it's default location, which ended up being:
/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin

Then I tried editing my .bashrc file, and added the following line to the end of the file: 
export PATH="/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin:$PATH"

I updated the .bashrc by using the following command: 
>$source ~/.bashrc

It still gives me the same error when I'm uploading the code. 

Am I doing something wrong? I could use some help. 

Thank you.

fpistm, 2020-05-18:
Well, as mentioned in #63  could you try to install Arduino IDE following official recommendations:
https://github.com/stm32duino/Arduino_Tools/issues/63#issuecomment-629561082

Diatrix, 2020-05-20:
Ok, I reinstalled the Arduino IDE not from the store, but from the file on the arduino.cc website and it works. 

For people having this sort of issue in the future: Make sure that the STM32CubeProgrammer is installed in the right spot. Sometimes it doesn't pick the correct default path. 
If it's installed on the right place, make sure that you didn't install the Arduino IDE from the store. Install it from the instructions as mentioned [here](https://www.arduino.cc/en/guide/linux). 

It should work fine after that! 

Thanks @fpistm for the help! :)

closed 2020-05-20 by Diatrix

Tomaszu97, 2020-06-07:
> Ok, I reinstalled the Arduino IDE not from the store, but from the file on the arduino.cc website and it works.
>
> For people having this sort of issue in the future: Make sure that the [...]

STM32CubeProgrammer picks /usr/local/ path when installed as root/ with sudo. Just run script **without** sudo.

fpistm, 2020-06-08:
> STM32CubeProgrammer picks /usr/local/ path when installed as root (or with sudo). Just run script **without** sudo.

Ah this can explain several thing. So, I guess I could add this path in the script.

Anyeos, 2021-01-03:
If someone can improve the script to support both locations ($HOME and /usr/local and /opt maybe too) it will be great because I don't want to install everything on my $HOME.

For now I tried with a symlink and it is working. So, another workaround can be make a symlink.
```
cd
ln -s /usr/local/STMicroelectronics STMicroelectronics
```

soheilpaper, 2021-06-14:
i have asked [here](https://askubuntu.com/questions/1345632/arduino-port-not-detected-and-can-not-run-it-as-root) and this part of [this](https://github.com/stm32duino/wiki/wiki/Upload-methods) document:
[image]

 would help you by running this command:

`export PATH="$HOME/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin:$PATH"
`