stm32duino/Arduino_Tools #65: cp a bin fails on Catalina

vsonneville-cherch, 2020-07-23:
Hi,

Realised when uploading from Arduino IDE on Mac OS X (10.15 Catalina), using MassSotrage method, it fails, saying that the `.ino.bin` is a directory.
Adding a `-r` to the `cp` command in `tools/macosx/massStorageCopy.sh` solved the issue, but I'm not sure it's the best way to do this.

Cheers

fpistm, 2020-08-03:
Hi,

could you be more precise?
Which Arduino IDE version? Core version?
Log ? (enable all the verbose output in the Arduino IDE preferences)

Thanks.

VVESTM, 2020-08-03:
Hi,

I have just made a test on latest versions (Arduino 1.8.13 and STM32core 1.9.0) with NucleoL476 board and mass storage upload method.
Result is OK : 
[image]

vsonneville-cherch, 2020-08-03:
Hi,
I was on Arduino 1.8.12, STM32Core 1.9.0 with a Nucleo F091RC and mass storage upload method.
I updated Arduino IDE to 1.8.13 and it fixed the issue.
Rolled back to 1.8.12 with STM32Core 1.9.0 and I enclose the verbose output here : [ErrorReport.txt](https://github.com/stm32duino/Arduino_Tools/files/5016631/ErrorReport.txt)
Given that an update of Arduino IDE fixed the issue, I don't know if it is still relevant, though.

fpistm, 2020-08-03:
Thanks for the quick feedback and tests.
It is linked to Arduino IDE and how it deals some paths.
I thinks it was solved in the 1.8.13 thanks https://github.com/arduino/arduino-cli/pull/690
I close this one as it is not relevant and not link to the tools.

closed 2020-11-11 by fpistm