stm32duino/Arduino_Tools #67: Java Runtime PATH problem under Windows (EXCEPTION_ACCESS_VIOLATION)

dadul96, 2020-08-30:
This problem occurs when a second (different) Java version is installed on a windows PC (besides the one provided with the Arduino IDE). Instead of using the provided Java version of the Arduino IDE the `maple_upload.bat`-file uses the Java version found in the environment path (oftentimes this is a different one).

This results in the following error message:
```
maple_loader v0.1
Resetting to bootloader via DTR pulse
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=12684, tid=0x0000000000003214
#
# JRE version: Java(TM) SE Runtime Environment (8.0_261-b12) (build 1.8.0_261-b12)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.261-b12 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\Daniel\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\1.4.0\tools\win\hs_err_pid12684.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

To solve this problem the following line should be added to the maple_upload.bat:
`if exist "C:\Program Files (x86)\Arduino\java\bin" set PATH=C:\Program Files (x86)\Arduino\java\bin;%PATH%`

This line temporarily changes the PATH during the execution of this batch file and ensures that the correct Java version is used.

This issue also has been discussed [here](https://www.stm32duino.com/viewtopic.php?t=604).

fpistm, 2021-04-15:
Hi @dadul96 
I never met this issue but it seems valid.
Anyway the proposed solution in #67 seems not complete  (See https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/846) nor compatible with other IDE (Ex: PIO or Eclispe +Sloeber) as they are using the Arduino Java.
My guess is that .jar should be replaced to avoid any further issue. This would be more efficient that trying to give correct (and older) java version. What will happen if next Arduino IDE version change the Java version? It will probably failed.

dadul96, 2021-04-15:
Hello @fpistm 
I see what you mean - my fix gives some nasty side effects to Linux and macOS users. Of course, the ideal solution would be to replace the .jar, but I don't think that I could lift such a huge architectural change. As a temporary solution, my old (unimproved) version of the PR (``if exist "C:\Program Files (x86)\Arduino\java\bin" set PATH=C:\Program Files (x86)\Arduino\java\bin;%PATH%``) could be used. This at least allows Arduino IDE users (who installed it in the default directory) to solve this issue. All other operating systems and IDE's would not be affected by this fix.

fpistm, 2021-04-15:
Right, anyway several user does not install Arduino to the default location.
I will replace the jar by a smaller binary doing the same than Linux and MacOS.

closed 2023-02-20 by fpistm

ChetW, 2023-03-18:
Thank you for this thread! Fixed my issue :)

PR #89 (2023-02-20): feat: unified maple_upload and dfu_util scripts (remove maple_loader.jar usage)
Fixes #67 

On windows, maple_loader.jar is replaced by small binary based on  http://sigrok.org/wiki/Libserialport.
Then dfu-utils is used.
This unified all host OS maple upload management as same scripts used for all.

Some fix made to scripts.

#### Tested:
| Host \ IDE | 1.8.19 | 2.0.3 |
| --- | --- | --- |
| windows | :heavy_check_mark: | :heavy_check_mark: | 
| Linux | :heavy_check_mark: | :heavy_check_mark: | 
| Mac | :heavy_check_mark: | :heavy_check_mark: | 

```
STMicroelectronics\tools\STM32Tools\2.1.1/w [... cut]
commit 6305aca0 (2023-02-20): feat: create upload_reset binary for windows

to remove java dependecy on Windows

fixes #67