stm32duino/Arduino_Tools #73: maple_loader.jar crash

dieter-l-git, 2021-02-12:
maple_loader.jar crashes when I try to upload a sketch to an maple mini clone (with bootloader 2.0) via DFU.
It doesn't matter which sketch - I used blink for testing. Same issue with Arduiono IDE and PlatformIO.
I using the current 1.9.0 release with Arduino 1.8.13 or PlatformIO 5.1.0 on Win10 20H2 with Java 15.0.27.
With Java 8.280 - same issue.

Errorlog attached. Seems that the jSSC-2.8_x86_64.dll is the issue.

I tried by myself to build a new maple_loader.jar from source with jssc2.9.2.jar- this is working without issues.

[hs_err_pid10232.log](https://github.com/stm32duino/Arduino_Core_STM32/files/5968047/hs_err_pid10232.log)

Thank you

fpistm, 2021-02-12:
I guess it is a duplicate of #67.
Never reproduced this issue. I think to remove the maple loader.

closed 2021-02-12 by fpistm

dieter-l-git, 2021-02-12:
Sorry for the duplicate. 
Setting the java path doesn't solve the issue. 

I will try on an other PC or Laptop in the next few days to see if I can reproduce the issue.

If you remove the maple loader - is there a suggestion what can be used to upload a sketch without additionally hardware needed?

fpistm, 2021-02-12:
You have to use the correct path. This is not necessarily the same one described in the issue than the one of your Arduino IDE path.

ArminJo, 2023-08-01:
The reason is this line in maple_upload.bat:  `java -jar maple_loader.jar %1 %2 %3 %4`
The java.exe is found by using the Windows %PATH% pointing to your java version15.
The `java `must be replaced by `......\arduino\java\bin\java` which points to the Arduino Java 1.8!
The arduino internal usage of `{the runtime.ide.path}/java/bin` does not work 😞 .

fpistm, 2023-08-01:
> The reason is this line in maple_upload.bat: `java -jar maple_loader.jar %1 %2 %3 %4` The java.exe is found by using the Windows %PATH% pointing to your java version15. The `java `must be replaced [...]

Now java is not used. No more issue.  See #89

ArminJo, 2023-08-01:
Thanks, you are right, I get confused with the 2 STM cores 😀