stm32duino/Arduino_Tools #78: Windows: EXCEPTION_ACCESS_VIOLATION on recent JRE on Windows 10

umaplehurst, 2021-07-07:
My colleague had issues with `maple_loader.jar` crashing out as per the JRE crash log snippet below but it looks like the root cause is due to an old version of JSSC. Replacing the bundled JSSC .jar file in Arduino_Tools to the 2.9.2 from upstream release sorted out the problem. Could the bundled JSSC .jar in Arduino_Tools be updated to the latest from upstream? Thanks!

Reference to JSSC issue: https://github.com/java-native/jssc/issues/65

```
Current thread (0x000001e363cc8000):  JavaThread "main" [_thread_in_native, id=8604, stack(0x0000000f57000000,0x0000000f57100000)]

Stack: [0x0000000f57000000,0x0000000f57100000],  sp=0x0000000f570fed00,  free space=1019k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [jSSC-2.8_x86_64.dll+0xb5db]

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J+0
j  jssc.SerialPort.openPort()Z+65
j  processing.app.Serial.<init>(Ljava/lang/String;ICIF)V+101
j  processing.app.Serial.<init>()V+40
j  CliTemplate.DFUUploader.emitResetPulse()V+10
j  CliTemplate.DFUUploader.uploadViaDFU(Ljava/lang/String;)Z+70
j  CliTemplate.CliMain.main([Ljava/lang/String;)V+104
v  ~StubRoutines::call_stub

siginfo: EXCEPTION_ACCESS_VIOLATION (0xc0000005), reading address 0x0000000061cb1bf3
```

fpistm, 2021-07-07:
Hi @umaplehurst 
Thanks for the report anyway it is a duplicate of https://github.com/stm32duino/Arduino_Tools/issues/67

closed 2021-07-07 by fpistm

umaplehurst, 2021-07-07:
@fpistm Ahhh, my apologies, I did see the other ticket but thought the `PATH` issue was an unrelated different problem. The stack trace from the DLL seems the same, so I think this is indeed a duplicate.