stm32duino/Arduino_Tools #79: 2.1.0/linux/hid-flash binary on Linux does not have execute bit set

bxparks, 2021-10-05:
**Describe the bug**
Execute bit is not set on the `hid-flash` binary. This causes flash upload to fail when using HID bootloader.

The permission mode for this file is currently:

```
$ cd /home/brian/dev/arduino-1.8.16/portable/packages/STMicroelectronics/tools/STM32Tools/
$ ls -l 2.1.0/linux/hid-flash
-rw-r--r-- 1 brian brian 42728 Sep 28 15:00 2.1.0/linux/hid-flash
```

**To Reproduce**

1. Install Arduino IDE 1.8.16 on Ubuntu 20.04.
2. STM32duino 2.1.0 in Arduino IDE.
3. Install HID Bootloader into STM32 blue pill (https://github.com/Serasidis/STM32_HID_Bootloader).
4. Select "Upload Method: HID Bootloader 2.2" in IDE menu.
5. Attempt to upload any sketch.

Result:
```
java.io.IOException: Cannot run program "/home/brian/dev/arduino-1.8.16/portable/packages/STMicroelectronics/tools/STM32Tools/2.1.0/linux/hid-flash": error=13, Permission denied
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2075)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=13, Permission denied
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch
```

**Expected behavior**

Compiled binary should upload properly.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.3 LTS
 - Arduino IDE version: 1.8.13, 1.8.16 
 - STM32 core version: 2.1.0
 - Tools menu settings if not the default: [...]
 - Upload method: HID Bootloader 2.2

**Board (please complete the following information):**
 - Name: STM32F103CB "blue pill"
 - Hardware Revision: [...]
 - Extra hardware used if any: [...]

**Additional context**

The same binary in 2.0.0 has the execute bit set:
```
$ $ cd /home/brian/dev/arduino-1.8.13/portable/packages/STMicroelectronics/tools/STM32Tools/
$ ls -l 2.0.0/linux/hid-flash
-rwxrwxr-x 1 brian brian 42728 Apr 14 15:00 2.0.0/linux/hid-flash
```

The problem is fixed if I manually set the execute-bit:
```
$ chmod +x 2.1.0/linux/hid-flash
```

fpistm, 2021-10-05:
Hi @bxparks 
I'm very surprised but you are right execute bit is missing and don't know why as it was tested when released.
Thanks for the report.

fpistm, 2021-10-05:
OK. after some investigation I've understood the issue.
Using git to check the file permission, it is ok
```
$ git ls-files  --stage hid-flash
100755 324cc30272e76018a3d00aeeaa528cce4b8f865e 0       hid-flash
```
On a Linux the file permissions in the local copy of the repo is correct but on Windows with git bash the file permissions is not correct and I don't know why. Only the hid_flash files are impacted (Linux and Mac).
As I've made the package under windows the file permission is not correct.

bxparks, 2021-10-05:
(Sorry, I did not know that there was a separate repo for the tools.)

It looks like the `upload-reset` binary in that directory is also affected:

```
$  ls -l upload-reset
-rw-r--r-- 1 brian brian 7946 Sep 28 15:00 upload-reset

$ git ls-files -s upload-reset 
100755 26985b857674d0c8d4999d53999a80f988591b96 0       upload-reset
```

You are right, I am not sure why the shell scripts are not affected.

fpistm, 2021-10-05:
Yes, I saw for upload-reset. 
I have to find why git bash do that. Probably, I will open an issue on git bash side.

fpistm, 2021-12-20:
I close this issue as it will be fixed with 2.1.1 release.

closed 2021-12-20 by fpistm