stm32duino/Arduino_Tools #82: [STM32CubeProgrammer] "-e all" does not work for STM32G031J6

olikraus, 2022-02-03:
Not sure where I can address the issue with the "-e all" commandline option for STM32CubeProgrammer.
So I put it here:

The "-e all" mass erase option does not work with the STM32G031J6. The following error is generated by the commandline version (Linux, USART interface):

```
Chip ID: 0x466 
BootLoader protocol version: 3.1
Device name : STM32G03x/STM32G04x
Flash size  : 64 KBytes (default)
Device type : MCU
Revision ID : --  
Device CPU  : Cortex-M0+

Mass erase ... 

Error: Mass erase operation failed.
```
Note: The J6 device only has 32KB Flash ROM.

See also #81

fpistm, 2022-02-04:
Hi @olikraus 
unfortunately, the script will not change this behavior. It only call the STM32Cube Programmer cli and options are valid as it properly call a mass erase when protocol is set to "1x".
Why the mass erase does not work is not link to the core nor the tools provided here. You can try with the STM32CubeProgrammer GUI to see if more info are available (probably flash protection issue) but honestly can't do anything here.
Same for the Flash size displayed, it it the Cube programmer which handle this.
Using STLink it displays 32kb while the UART BL displays 64k (default).
Probably the BL try a mass erase on 64kb in Serial that's why it failed. But it is handle by the STM32cube Programmer.

closed 2022-02-04 by fpistm

fpistm, 2022-02-04:
Duplicate of #81

olikraus, 2022-02-04:
> GUI to see if more info are available (probably flash protection issue) but honestly can't do anything here.

I really like the integration of CubeProgrammer into Arduino IDE. It reduces the effort the Arduino users.

> Using STLink it displays 32kb while the UART BL displays 64k (default).

Yes correct, this is another issue what I observed (and maybe also the root cause why the mass erase fails)

>  but honestly can't do anything here.

My hope was, that you could raise a bug request for STM32CubeProgrammer. I assume that you are part of ST and maybe you have access to the internal bug tracker for STM32CubeProgrammer. Or, if such a public place is available, where I can report issues with STM32CubeProgrammer I will also do so, if you tell me where to create such a bug request.

> Duplicate of https://github.com/stm32duino/Arduino_Tools/issues/81

Sort of. 81 was for the .sh script and 82 (this issue) is for cube programmer.

But I understand, this issue can not be solved within this git project.
Nevertheless, I think cubeprogrammer has some bugs related the STM32G0 device, but I do not know where to address this.

Please close this issue.

olikraus, 2022-02-04:
Nevertheless, let me create a small of issues, we figured out for STM32CubeProgrammer related to the STM32G031J6 device

1. "-e all" creates an error
2. "-e [0 31] does NOT create an error (expected valid range should be 0..15 because one page has 2kb)
3. Flash size is not displayed correctly

fpistm, 2022-02-04:
> My hope was, that you could raise a bug request for STM32CubeProgrammer. I assume that you are part of ST and maybe you have access to the internal bug tracker for STM32CubeProgrammer. Or, if such a [...]

Right, this will be the case when the first issue you open will be finish to investigate :) I close it here because it does not rely on the core nor the tools provided here.
So of course we will probably open an internal bug tracker about those issues.