stm32duino/Arduino_Tools #83: In latest ArduinoIDE 1.8.19, `stm32CubeProg.sh` claims `STM_Programmer.sh` is not on my `PATH`.

zfields, 2022-03-11:
STM_Programmer.sh is added to my `PATH` in `~/.bashrc`. It works fine in the Arduino Extension in VSCode, and I can execute the logic manually in a terminal application.

Maybe the logic needs to be revisited for the latest Arduino IDE? :shrug:

fpistm, 2022-03-11:
Hi @zfields 
I don't understand your point about the latest Arduino IDE.
I think you have installed IDE thanks the package manager of your Linux distribution and so it is executed in a specific env. Some user already raised this issue but honestly I have no way to find a proper fix.

This is explain in the wiki:
https://github.com/stm32duino/wiki/wiki/Upload-methods#requirement

[image]

zfields, 2022-03-11:
I did not install via `snap`. I downloaded the `.tar.gz` file directly from `https://www.arduino.cc/en/software`. I extracted the file into the system at `/opt/arduino`, then I ran the installation script (`arduino-linux-setup.sh`).

This should be a very common workflow for any linux user.

zfields, 2022-03-11:
No, I do not launch Arduino from the command line.

After running `arduino-linux-setup.sh`, it is installed into the system as an Application.

To launch it, I press the "Windows Key" and type Arduino.

I will try to launch Arduino from the command line, and see if that makes the difference.

fpistm, 2022-03-11:
OK that explain why your .bashrc PATH is not used.

fpistm, 2022-03-11:
For the desktop icon you could probably set the PATH in its command line. Something like this (not tested):

`Exec=sh -c "PATH=/opt/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin:$PATH arduino"`

or easiest way, simply install the STM32CubeProgrammer to the default directory.

zfields, 2022-03-12:
I backed into this by updating the script `~/.arduino15/packages/STMicroelectronics/tools/STM32Tools/2.1.1/stm32CubeProg.sh`.

I added 

```    if ! command -v $STM32CP_CLI >/dev/null 2>&1; then
      export PATH="/opt/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin":$PATH
    fi
```
so it would find `STM32_Programmer.sh`

fpistm, 2022-03-12:
Hi @zfields 
Of course but I could not add all user path in the script.
What about launching arduino from the command line? Did you try to update your desktop icon? ....
How did you install STM32CUBEPROGRAMMER?
Please answer all questions.

zfields, 2022-03-12:
Sorry for the delay. The latest Arduino IDE has been trashing my option bytes, so I have to try and restore the Swan (by manually setting the option bytes) using the `STM32CubeProgrammer`. It's really slow going. I've even been resetting my machine to make sure the tools aren't in a bad place.

zfields, 2022-03-12:
I can confirm that when called from the command line the Arduino IDE works as expected without the modification to `stm32CubeProg.sh`.

As for the STM32CubeProgrammer, I downloaded the `.zip` from ST, extracted it, ran `sudo SetupSTM32CubeProgrammer-2.10.0.linux`, and chose `/opt/STMicroelectronics/STM32Cube/STM32CubeProgrammer` as the install location.

fpistm, 2022-03-12:
> chose `/opt/STMicroelectronics/STM32Cube/STM32CubeProgrammer` as the install location.

It is proposed by the installer or  you decided to set this path ?

fpistm, 2022-03-12:
> , I have a command launcher which is similar to the "Start" menu.

So you should be able to update it to set the PATH

zfields, 2022-03-12:
The installer proposes, `~/STMicroelectronics/STM32Cube/STM32CubeProgrammer`, I just changed it from `~` to `/opt`.

zfields, 2022-03-12:
It doesn't work exactly like Windows, so it's not a shortcut I can modify so easily. It uses a `.desktop` file:

```none
[Desktop Entry]
Type=Application
Name=Arduino IDE
GenericName=Arduino IDE
Comment=Open-source electronics prototyping platform
Exec="/opt/arduino/arduino-1.8.19/arduino"
Icon=arduino-arduinoide
Terminal=false
Categories=Development;IDE;Electronics;
MimeType=text/x-arduino;
Keywords=embedded electronics;electronics;avr;microcontroller;
StartupWMClass=processing-app-Base
```

If you know how to update the file accordingly, I'll happily give it a try.

fpistm, 2022-03-12:
You forgot the "sh -c"

And do not hesitate to google.:wink:

zfields, 2022-03-13:
It didn't work the way I was expecting

```
Exec="sh -c \"PATH=/opt/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin/ /opt/arduino/arduino-1.8.19/arduino\""
```

Perhaps you could share the link...

fpistm, 2022-03-14:
Maybe the $PATH is missing.

Exec="sh -c \"PATH=/opt/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin/:\$PATH /opt/arduino/arduino-1.8.19/arduino\""

Ref:
https://specifications.freedesktop.org/desktop-entry-spec/latest/ar01s07.html

fpistm, 2022-03-14:
> I don't really think your suggestion could technically work. Honestly, it doesn't really matter, because we are squarely in the territory of this being too much trouble to be worth it.

I know I have already do some things like this and it is not linked to this repo. It is only user OS config.

> From now on, I will just "launch it from the command line" if I plan to use an ST board. In the meanwhile, it would be good if you could revisit your logic (or installer) to make this happen automatically.

Unfortunately, I have no solution.
I can't package and deliver the STM32CubeProgrammer. 
There is no other generic solution to provide the same support level of upload method for all the series.
So the only way is that user have to install the STM32Cubeprogrammer on his side.
As stated if you had installed the STM32CubeProgrammer in the default path, it will work by default. As you changed the path it is up to you to set properly the PATH as stated by the script error.

closed 2022-03-14 by fpistm

fpistm, 2022-03-15:
Hi @zfields ,
I use the install.sh. So in fact it is only to create a desktop file 🙄 
I've remove the STM32CubeProgrammer from the default path and then I've update the desktop entry file like this and it works like a charm.

`Exec=sh -c "PATH=/opt/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin:$PATH /opt/arduino/arduino-1.8.19/arduino"`

So to conclude it is up to end user to properly configure his desktop entry to provide the correct env if the STM32CubeProgrammer is not installed in the default path.