stm32duino/Arduino_Tools #85: stm32duino.com - impossible to read or login (Cookie Acceptance Required - no way to accept)

Andy2No, 2022-07-11:
This probably isn't the best place to ask, but I don't know how else to ask it.  I'm unable to login to stm32duino.com, search or even read a page as a guest.  All I get is a notice that says "Cookie Acceptance Required", but I see no way to accept cookies, despite disabling all forms of blocking and tracking protection.

This also meant I couldn't use the Contact link, at the bottom of the login page, to ask you about it.

This is what it says when I try to login:

```
Cookie Acceptance Required

You must accept the Cookie Policy of Arduino for STM32 before you can either register on this site or, if you are already registered, before you can login to the site.
```

This is what it says, in a dialog box, when I click any link on the site, including "Contact us"

```
You cannot access links on this board until you have accepted the Cookie Policy.
```

Is the forum dead?  If so, a notice saying so would be helpful, but I see there is still a link to it near the top of the github page here:

https://github.com/stm32duino

fpistm, 2022-07-11:
Hi @Andy2No 
Try to start from the main page. You should have a banner which request to accept.
Maybe clean up your cache.

closed 2022-07-11 by fpistm

fpistm, 2022-07-11:
Fine. Thanks for the feedback.