stm32duino/Arduino_Tools #91: 2.2.0/linux/hid-flash binary on Linux does not have execute bit set

bxparks, 2023-05-23:
Hi,
I just upgraded to STM32duino 2.5.0, and I encountered the same problem as bug #79 that I filed 1.5 years ago: various binaries under the `STMicroelectronics/tools/STM32Tools/2.2.0/linux` do not have their eXecute bit set, in particular, the `hid-flash`. This causes the Arduino IDE to fail with a Java exception, preventing flashing of my STM32 Blue Pill using the HID bootloader.

```
$ cd .../STMicroelectronics/tools/STM32Tools/2.2.0/linux
$ ls -l 
-rw-r--r-- 1 brian brian   411 Mar 28 15:00 45-maple.rules
drwxr-xr-x 2 brian brian  4096 Mar 28 15:00 dfu-util/
drwxr-xr-x 2 brian brian  4096 Mar 28 15:00 dfu-util_x86_64/
-rw-r--r-- 1 brian brian 42728 Mar 28 15:00 hid-flash
-rwxr-xr-x 1 brian brian   596 Mar 28 15:00 install.sh*
-rwxr-xr-x 1 brian brian  2414 Mar 28 15:00 massStorageCopy.sh*
-rw-r--r-- 1 brian brian  7946 Mar 28 15:00 upload_reset

$ ls -l dfu-util
-rw-r--r-- 1 brian brian  31663 Mar 28 15:00 dfu-prefix
-rw-r--r-- 1 brian brian  31962 Mar 28 15:00 dfu-suffix
-rw-r--r-- 1 brian brian 115520 Mar 28 15:00 dfu-util

$ ls -l dfu-util_x86_64/
-rw-r--r-- 1 brian brian  47032 Mar 28 15:00 dfu-prefix
-rw-r--r-- 1 brian brian  47088 Mar 28 15:00 dfu-suffix
-rw-r--r-- 1 brian brian 151688 Mar 28 15:00 dfu-util
```

The workaround is to manually set this bit:
```
$ chmod +x hid-flash upload_reset dfu-util/* dfu-util_x86_64/*
```

The Java exception is:
```
/home/brian/Downloads/arduino/ArduinoIDE/arduino-1.8.19/portable/packages/STMicroelectronics/tools/STM32Tools/2.2.0/linux/hid-flash /tmp/arduino_build_427209/AutoBenchmark.ino.bin ttyACM0 
java.io.IOException: Cannot run program "/home/brian/Downloads/arduino/ArduinoIDE/arduino-1.8.19/portable/packages/STMicroelectronics/tools/STM32Tools/2.2.0/linux/hid-flash": error=13, Permission denied
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2081)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=13, Permission denied
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch
```

fpistm, 2023-05-24:
Hi @bxparks 
I've forgot this and generate again under git bash. 
Sorry, I will properly regenerate within the next release in the 2.2.1 packages.

bxparks, 2023-05-24:
@fpistm : thank you for the quick attention on this, and all your work on STM32duino

fpistm, 2023-07-05:
I close it as I will release the 2.2.1

closed 2023-07-05 by fpistm