stm32duino/Arduino_Tools #99: Cannot upload sketch on MacOS

miroslavpetrov, 2024-07-01:
**Describe the bug**
I am also not able to upload sketches from the Arduino IDE on MacOS. I have STM32cubeprog installed and working on my Mac but thats what I get from the Arduino IDE:

```
Sketch uses 110728 bytes (42%) of program storage space. Maximum is 262144 bytes.
Global variables use 8648 bytes (13%) of dynamic memory, leaving 56888 bytes for local variables. Maximum is 65536 bytes.
getopt: illegal option -- o
Terminating...
Failed uploading: uploading error: exit status 1
```

**To Reproduce**
Steps to reproduce the behavior:
No matter what upload method I select - SWD, UART, DFU the output is the same.
I am only able to upload the sketch by exporting compiled binary, open sketch folder in finder and move the .bin file to STLink drive.

**Expected behavior**
I expect to be able to flash STM32WL using Arduino IDE

**Desktop**
 - OS: Mac OS Sonoma 14.5
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8.0
 - STM32 Low Power: 1.2.5
 - STM32 LoRaWAN: 0.2.0
 - Tools menu settings if not the default: Core logs enabled, Optimize: smallest, Runtime Newlib nano, UART: Enabled(generic serial), Board part number: Lora-E5 mini
 - Upload method: SWD
 
**Board (please complete the following information):**
 - Name: Seeed Wio-E5 Mini based on STM32wle5jc
 - Hardware Revision: v1.0

fpistm, 2024-07-01:
Hi @miroslavpetrov 
I've found the issue.
Please, Could you try the fix provided within #99?

miroslavpetrov, 2024-07-01:
@fpistm Thank you. Now it works like a charm. I am closing the issue. 

```
Sketch uses 128352 bytes (48%) of program storage space. Maximum is 262144 bytes.
Global variables use 8652 bytes (13%) of dynamic memory, leaving 56884 bytes for local variables. Maximum is 65536 bytes.
Selected interface: swd
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.14.0                  
      -------------------------------------------------------------------

ST-LINK SN  : 0032001B3431511837393330
ST-LINK FW  : V3J12M3
Board       : STLINK-V3MINIE
Voltage     : 3.30V
SWD freq    : 8000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x497
Revision ID : Rev Z
Device name : STM32WLxx
Flash size  : 256 KBytes
Device type : MCU
Device CPU  : Cortex-M4
BL Version  : 0xC3

Memory Programming ...
Opening and parsing file: LowPowerBasic.ino.bin
  File          : LowPowerBasic.ino.bin
  Size          : 125.71 KB 
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 62]
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:03.323

RUNNING Program ... 
  Address:      : 0x8000000
Application is running, Please Hold on...
Start operation achieved successfully
```

fpistm, 2024-07-02:
Hi @miroslavpetrov 
Thanks for the test. I reopen the issue as the PR is not merged. I have some other concern about BSD getopt limitation (when arguments have space) and I need to validate under windows.
Issue will be closed automatically when the PR will be merged.

PR #100 (2024-07-17): fix(stm32CubeProg): fallback to BSD getopt on mac
By default, MacOS uses getopt from FreeBSD and not the GNU-based one.
Then "-o" and long options are not supported except if gnu-getopt is installed.

Fixes #99

Main advantage to kept GNU getopt by default is it is able to deal with space in arguments.
Install the package `gnu-getopt` via e.g. [Homebrew](https://brew.sh):

```bash
brew install gnu-getopt
```
If not installed the BSD getopt is used but long option and space in arguments are not supported.
commit 8de264f8 (2024-07-17): fix(stm32CubeProg): fallback to BSD getopt for MacOS

By default, MacOS uses getopt from FreeBSD and not the GNU-based one.
Then "-o" and long options are not supported except if gnu-getopt
is installed.

Fixes #99
closed 2024-07-17 by fpistm